ethz-asl/maplab #280: Import custom format of 3D map (e.g., COLMAP) to maplab

dadung, 2020-09-20:
Hello maplab team,

I have a 3D map built by images only (using [COLMAP](https://colmap.github.io/)), is it possible for me to import a custom 3D map format (e.g., COLMAP format in my case) to maplab format? I would like to use "Map summarization" of maplab to compress my map.

Thank you!

youzhenxing, 2021-03-20:
I have the same need，is there any docs for maplab map format like [colmap sparse format](https://colmap.github.io/faq.html#reconstruct-sparse-dense-model-from-known-camera-poses)?  so that we can covert my maps as malab's foramt to use maplab excellent tools like map merging and summary.

youzhenxing, 2021-03-20:
I have the same  problem，is there any docs for maplab map format like [colmap sparse format](https://colmap.github.io/faq.html#reconstruct-sparse-dense-model-from-known-camera-poses)? so that we can covert my maps as malab's foramt to use maplab excellent tools like map merging and summary.

GoroYeh-HRI, 2024-05-29:
Hi, 

@dzungdoan6  and @youzhenxing  I have the same problem here, do you have any updates on how to set up the interfacing? Thank you!