ethz-asl/maplab #295: failed to rosrun rovioli tutorial_euroc

sensationHJ, 2021-02-18:
**TODO:** ADD YOUR QUESTION/PROBLEM DESCRIPTION HERE
Hi 
I'm doing maplab tutorial and when i run this ,

source ~/maplab_ws/devel/setup.bash
roscore&
rosrun rovioli tutorial_euroc save_folder MH_01_easy.bag

i can see this ERROR

ERROR: unknown command line flag 'imu_parameters_maplab'
ERROR: unknown command line flag 'imu_parameters_rovio'
ERROR: unknown command line flag 'ncamera_calibration'

[image]

i cant figure out how can i do...

plz help me...!

i followed the procesure of 

https://maplab.asl.ethz.ch/docs/devel/dostreif/pages/installation/A_Installation-Ubuntu.html

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System : 64bit ubuntu 18.04
     - Eigen : 3.3.4
     - ROS : melodic
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

Gian-Alessio, 2021-05-14:
Could you solve the error?

smauq, 2021-06-03:
Sorry for the late answer but I think you have the same issue as here `https://github.com/ethz-asl/maplab/issues/315`. Basically for now I would suggest switching away from the `develop` branch to one of the stable release branches until we can update the tutorials