ethz-asl/maplab #312: Rovioli: imu_parameters_maplab and ncamera_calibration do not exist?

Gian-Alessio, 2021-05-15:
OS: 18.04
Eigen: 3.3.4
ROS: Melodic
I tried to run maplab on 16.04 and 18.04 and got the same problem twice. While installing I followed the docs and I am building from the /develop branch. 
My problem is I cannot run rovioli as mentioned in the doc. When i try to run: `rosrun rovioli tutorial_euroc` I get the following error on both tested systems:
`ERROR: unknown command line flag 'imu_parameters_maplab'`
`ERROR: unknown command line flag 'ncamera_calibration'`

**were these parameters deleted? **

When I run: `rosrun rovioli rovioli --help`, I cannot find them? 
Is there a new way to pass the cailbration files? if yes how?

smauq, 2021-06-03:
The develop branch is still work in progress. We are changing a lot of things and are still in the process of documenting it and changing the tutorials (soon to come). In the mean time you should use one of the official release branches if you want to easily be able to follow the tutorials e.g. I would suggest`https://github.com/ethz-asl/maplab/tree/pre_release_public/july-2018`

BystanderJ, 2023-02-09:
Same problems here. And the solution is :
First, # these 2 lines like below:
# --ncamera_calibration=$NCAMERA_CALIBRATION  
# --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB 

then run the script it will reminds you that no calibration_file has been found.
So add the command: "--sensor_calibration_file=$NCAMERA_CALIBRATION"

And the calibration file must contain imu, so change the ncamera.yaml to euroc-mono.yaml like below:
NCAMERA_CALIBRATION="$ROVIO_CONFIG_DIR/euroc-mono.yaml" 

Then the script can work.