ethz-asl/maplab #315: rosrun rovioli tutorial_euroc doesn't work.

wendaoyuchen, 2021-06-02:
it really makes me confused. when I run 
rosrun rovioli tutorial_euroc /home/xxxx/result/ /home/xxxx/workspace/dataset/TumDataset/MH_01_easy.bag

it told me 
ERROR: unknown command line flag 'datasource_rosbag'
ERROR: unknown command line flag 'datasource_type'
ERROR: unknown command line flag 'imu_parameters_maplab'
ERROR: unknown command line flag 'imu_parameters_rovio'
ERROR: unknown command line flag 'map_builder_save_image_as_resources'
ERROR: unknown command line flag 'ncamera_calibration'
ERROR: unknown command line flag 'optimize_map_to_localization_map'
ERROR: unknown command line flag 'save_map_folder'

Is there anyone can help me?

wendaoyuchen, 2021-06-03:
so finally, nobody answer in maplab.

smauq, 2021-06-03:
Well I mean did you source your compiled workspace? I.e. `source catkin_ws/devel/setup.bash` or whatever the path is on your machine

wendaoyuchen, 2021-06-04:
thanks for your apply. I didi source it

cd /home/liuzhen/catkin_maplab/build/rovioli; catkin build --get-env rovioli | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................................................................................................................................................................................
Finished  <<< rovioli                                         [ 1 minute and 35.4 seconds ]                                                                                                                                                                                       
[build] Summary: All 91 packages succeeded!                                                                                                                                                                                                                                       
[build]   Ignored:   14 packages were skipped or are blacklisted.                                                                                                                                                                                                                 
[build]   Warnings:  7 packages succeeded with warnings.                                                                                                                                                                                                                          
[build]   Abandoned: None.                                                                                                                                                                                                                                                        
[build]   Failed:    None.                                                                                                                                                                                                                                                        
[build] Runtime: 5 minutes and 49.7 seconds total.                                                                                                                                                                                                                                
liuzhen@liuzhen-ThinkStation-P320:~/catkin_maplab$ source devel/setup.bash 
liuzhen@liuzhen-ThinkStation-P320:~/catkin_maplab$ rosrun rovioli 
rovioli                      run_rovioli                  run_rovioli_live             tutorial_euroc               tutorial_euroc_live          tutorial_euroc_localization  
liuzhen@liuzhen-ThinkStation-P320:~/catkin_maplab$ rosrun rovioli tutorial_euroc
ERROR: unknown command line flag 'datasource_rosbag'
ERROR: unknown command line flag 'datasource_type'
ERROR: unknown command line flag 'imu_parameters_maplab'
ERROR: unknown command line flag 'imu_parameters_rovio'
ERROR: unknown command line flag 'map_builder_save_image_as_resources'
ERROR: unknown command line flag 'ncamera_calibration'
ERROR: unknown command line flag 'optimize_map_to_localization_map'
ERROR: unknown command line flag 'save_map_folder'
liuzhen@liuzhen-ThinkStation-P320:~/catkin_maplab$

smauq, 2021-06-04:
What branch of maplab are you on? If you want to be able to follow the turorial you need to be on master or one of the release branches (see the same issue here https://github.com/ethz-asl/maplab/issues/312)

BystanderJ, 2023-02-09:
Same problems here. And the solution is :
First, # these 2 lines like below:
# --ncamera_calibration=$NCAMERA_CALIBRATION  
# --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB 

then run the script it will reminds you that no calibration_file has been found.
So add the command: "--sensor_calibration_file=$NCAMERA_CALIBRATION"

And the calibration file must contain imu, so change the ncamera.yaml to euroc-mono.yaml like below:
NCAMERA_CALIBRATION="$ROVIO_CONFIG_DIR/euroc-mono.yaml" 

Then the script can work.

ckq1234, 2023-02-20:
> Same problems here. And the solution is : First, # these 2 lines like below:
>
> # --ncamera_calibration=$NCAMERA_CALIBRATION
> # --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB
> then run the script it [...]
tk