ethz-asl/maplab #316: provide loop information from other algorithm

CanCanZeng, 2021-06-08:
Hi, is there any  way to provide loop closing information to maplab? I find that in some difficult case, use default descriptor will fail in closing loop, so I wonder if I can use other way to find loop (like sift or superpoint), or just give good candidates using netvlad.

dymczykm, 2021-06-09:
@CanCanZeng thanks for your question. What you propose is definitely feasible, you would need however to interface with the following modules:
- [LoopDetectorNode](https://github.com/ethz-asl/maplab/blob/master/algorithms/loopclosure/loop-closure-handler/include/loop-closure-handler/loop-detector-node.h) (the high level node that queries for global localization)
- [LoopDetectorHandler](https://github.com/ethz-asl/maplab/blob/master/algorithms/loopclosure/loop-closure-handler/include/loop-closure-handler/loop-closure-handler.h) (the module that summarizes the matches and runs RANSAC geometric verification on the results + merges the landmarks)

Hope that helps!

CanCanZeng, 2021-06-09:
Thank you for your help! I will look into these code.