ethz-asl/maplab #320: No Loop clsoure edges found

Gian-Alessio, 2021-06-23:
Hey guys! 

Using maplab(devel branch) I cannot get any Loop Closures, or more precised the pose graph relaxation does not work. I tried it with different data, including the euroc data from the maplab tutorial and I always get the following message:
```
I0623 17:35:48.983312 10682 loop-detector-node.cc:1032] 
Loop closure result:
 - missions in database: 48fe..0000, 
 - localized 661/1838 vertices
I0623 17:35:49.124603 10682 vi-map-relaxation.cc:92] No loop closure edges found, relaxation cannot proceed.
```
why can't loop closure edges be found? 
I have also tried to perform bundle adjustments before, but it still does not work!

smauq, 2021-06-26:
Can you specify a bit more precisely what steps you took and what command sequence you are doing? Because from the limited  output I see you do have 661 vertices localized.

Gian-Alessio, 2021-06-26:
```
maplab <>:/$ load -map_folder save_map/map_Euroc_MH01/
Progress: 0: 50.0% [#########] 1:  0.0% [---------] 2:  0.0% [---------] 3:  0.0Progress: 0:100.0%  1:  0.0%  2:  0.0%  3: 50.0%  4:  0.0%  5:  0.0%  6:  0.0%  Progress: 0:100.0%  1: 50.0%  2: 50.0%  3:100.0%  4: 50.0%  5:  0.0%  6: 50.0%  Progress: 0:100.0%  1:100.0%  2: 50.0%  3:100.0%  4: 50.0%  5:  0.0%  6: 50.0%  Progress: 0:100.0%  1:100.0%  2:100.0%  3:100.0%  4: 50.0%  5:  0.0%  6: 50.0%  Progress: 0:100.0%  1:100.0%  2:100.0%  3:100.0%  4:100.0%  5:  0.0%  6: 50.0%  Progress: 0:100.0%  1:100.0%  2:100.0%  3:100.0%  4:100.0%  5:  0.0%  6:100.0%  Progress: 0:100.0%  1:100.0%  2:100.0%  3:100.0%  4:100.0%  5:100.0%  6:100.0%                      
--->Complete!
I0626 17:18:41.926838  9216 vi-map-serialization.cc:362] Loaded VIMap from "save_map/map_Euroc_MH01/".
maplab <map_Euroc_MH01>:/$ relax
I0626 17:18:58.629611  9216 loop-detector-node.cc:1032] 
Loop closure result:
 - missions in database: 48fe..0000, 
 - localized 674/1838 vertices
I0626 17:18:58.770757  9216 vi-map-relaxation.cc:92] No loop closure edges found, relaxation cannot proceed.
```

@smauq 
I just load the map into the maplab console and then I run "relax". So that the relaxation will correct for cases where large drift occurred, but somehow only landmarks are merged and the Pose Graph Relaxation it seems to fail.

smauq, 2021-06-30:
I had to recreate it quickly myself, but your issue in this case is the parameter `lc_edge_min_distance_meters` which has a default value of 1.0m. This means that a loop closure edge won't be added if the distance between the two vertices is smaller than that threshold (because we consider that for very small distances merging landmarks is more precise and lc edges are used to bridge bigger gaps as you also said). The issue here is that the map is relatively small and the odometry didn't really drift that much.

Anyways if you set the flag to a lower value it works as expected
```
relax --lc_edge_min_distance_meters=0.1
I0630 20:41:21.493264  7393 loop-detector-node.cc:1032] 
Loop closure result:
 - missions in database: 0c29..0000, 
 - localized 668/1837 vertices
I0630 20:41:21.576434  7393 vi-map-relaxation.cc:95] 30 loopclosure edges found and TEMPORARILY added to the map.
...
```
Just so you know I debugged this by just increasing verbosity on the relax command using `-v=10` and saw from the log output what was going on. Just for future reference as a tip.