ethz-asl/maplab #321: ROVIOLI in VIO mode runs very slow

sumitsarkar1, 2021-07-01:
I have successfully installed master branch of maplab, but the ROVIOLI in VIO example runs very slow with MH_01_easy.bag. I checked the Hz of rosbag topics which are fine with imu at 200 Hz and cams at 20 Hz. However the tutorial example runs very slow as I see in the "Tracker0" window the images are updated once or twice in 2 seconds and I get timeout and other warnings. Please help. I have strictly followed most of the installation and tutorial guidelines.

[image]
[image]
[image]

smauq, 2021-07-01:
Can you say exactly how you launch everything? And just a few details about your setup, like are you trying to run it through ssh, in a docker etc?

sumitsarkar1, 2021-07-01:
1.) Platform : i7 , 8GB RAM, Ubuntu 16 , ROS Kinetic
2.) No ssh , no docker, just a laptop
3.) Then I follow the instructions at https://github.com/ethz-asl/maplab/wiki/Running-ROVIOLI-in-VIO-mode

smauq, 2021-07-01:
Can I suggest two things to see if any help. Instead of master branch take the newest release candidate, i.e. this https://github.com/ethz-asl/maplab/tree/pre_release_public/july-2018 and try again. And also while running ROVIOLI can you take a screenshot of htop to see the CPU usage?

sumitsarkar1, 2021-07-02:
It got solved with the master branch...I was tampering with the launch file...my bad..thanx a lot for the response

closed 2021-07-02 by sumitsarkar1