ethz-asl/maplab #346: No valid coordinate data!???

Pan-long-ming, 2021-11-14:
When i run rovioli using my own image data,it occurs error-"    ERROR: No valid coordinate data!",Why this could happen,what it means?

smauq, 2021-11-19:
No idea, haven't seen this before. The error comes deep from within rovio. I would first double-check everything (calibration, topics, etc.). Then maybe you have something specific and weird, like an extremely tiny resolution image might not work?

Pan-long-ming, 2021-11-19:
Thank you, I know that is the rovio's problem.

closed 2021-11-19 by Pan-long-ming

ShrutheeshIR, 2022-04-23:
Hi @Pan-long-ming , I face the same issue with the TUM VI dataset, were you able to fix it?

B1ueber2y, 2024-10-29:
Helloo, facing the same issue. Were you able to fix it? @Pan-long-ming @ShrutheeshIR