ethz-asl/maplab #379: help!

a2801603396, 2023-03-25:
Can someone execute this code for me and then compress the file and send it to this email "578646910@qq.com" , thanks!

code:"git clone git@github.com:ethz-asl/maplab.git --recursive -b develop"

smauq, 2023-03-28:
You can find updated install instructions here:
- [Ubuntu 18.04](https://github.com/ethz-asl/maplab/blob/e84e3a990c1a812ae7ab776a4d58049432cc4a5f/docs/pages/installation/A_Installation-Ubuntu-18.04.md``)
- [Ubuntu 20.04](https://github.com/ethz-asl/maplab/blob/e84e3a990c1a812ae7ab776a4d58049432cc4a5f/docs/pages/installation/A_Installation-Ubuntu-20.04.md)

Essentially doing `git clone git@github.com:ethz-asl/maplab.git --recursive` should be fine. If you want an older version of the code you can check out the releases [here](https://github.com/ethz-asl/maplab/releases).

MF-Ahmed, 2023-04-26:
Hi I am having the same error on Ubunto 20.04, ROS Noetic 
git clone git@github.com:ethz-asl/maplab.git --recursive                                                                                                         catkin_ws/src
Cloning into 'maplab'...
Warning: Permanently added the ECDSA host key for IP address 'xxx.xx.xxx' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

closed 2023-04-26 by a2801603396

MF-Ahmed, 2023-04-26:
Hi I am following the installation instructions  https://github.com/ethz-asl/maplab/blob/e84e3a990c1a812ae7ab776a4d58049432cc4a5f/docs/pages/installation/A_Installation-Ubuntu-20.04.md 
and installed all the required dependencies  but I'm still not able to git clone

a2801603396, 2023-04-26:
> Hi I am following the installation instructions https://github.com/ethz-asl/maplab/blob/e84e3a990c1a812ae7ab776a4d58049432cc4a5f/docs/pages/installation/A_Installation-Ubuntu-20.04.md and installed all the required dependencies but I'm still not able to git clone

Hi，I gave up using the git clone method to download the code, and instead obtained the code using the downloaded zip compression package, where the sub modules were also obtained by downloading the zip compression package. After decompressing all sub modules and placing them in the corresponding directory, the compilation will be fine.

MF-Ahmed, 2023-04-26:
Thanks for the info, I have downloaded the Maplab 1.3 source code .zip file but its README.md says its for Ubunto 14.04+ROS indigo or Ubunto 16.04+ROS Kinetic ...

a2801603396, 2023-04-26:
> Thanks for the info, I have downloaded the Maplab 1.3 source code .zip file but its README.md says its for Ubunto 14.04+ROS indigo or Ubunto 16.04+ROS Kinetic ...

The document states that maplab2.0 is applicable to ubuntu20.04 and 18.04, while yours should only be able to use maplab2.0. If you want to use maplab1.3, you may need to install ubuntu14.04 or 16.04.

MF-Ahmed, 2023-04-26:
Thanks for the info, I have downloaded the Master branch (which i think should be Maplab 2.0 ) as a zip file from https://github.com/ethz-asl/maplab, uncompressed it in my catkin_ws/src folder .. catkin  build gives me the following error:

 
Errors     << maplab_common:cmake /home/usr/data/catkin_ws/logs/maplab_common/build.cmake.001.log                                             
CMake Warning (dev) at CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /home/usr/data/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findaslam_cv_common.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "aslam_cv_common", but CMake did not find one.

  Could not find a package configuration file provided by "aslam_cv_common"
  with any of the following names:

    aslam_cv_commonConfig.cmake
    aslam_cv_common-config.cmake

  Add the installation prefix of "aslam_cv_common" to CMAKE_PREFIX_PATH or
  set "aslam_cv_common_DIR" to a directory containing one of the above files.
  If "aslam_cv_common" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:5 (catkin_simple)

a2801603396, 2023-04-26:
> 感谢您提供的信息，我已经从 https://github.com/ethz-asl/maplab 下载了主分支（我认为应该是 Maplab 2.0）作为 zip 文件，在我的 catkin_ws/src 文件夹中解压缩它..Catkin 构建给了我以下错误：
> 
> 错误 << maplab_common：cmake /home/usr/data/catkin_ws/logs/maplab_common/build.cmake.001.log CMakeLists.txt：2 （项目） 上的 CMake 警告 （dev）：未设置策略 CMP0048：project（） 命令管理 VERSION 变量。运行“cmake --help-policy cmp0048”了解策略详细信息。使用 cmake_policy 命令设置策略并禁止显示此警告。
> 
> 以下变量将设置为空：
> 
> ```
> CMAKE_PROJECT_VERSION
> CMAKE_PROJECT_VERSION_MAJOR
> CMAKE_PROJECT_VERSION_MINOR
> CMAKE_PROJECT_VERSION_PATCH
> ```
> 
> 此警告适用于项目开发人员。使用 -Wno-dev 来抑制它。
> 
> CMake Error at /home/usr/data/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake：38 （find_package）：由于没有在CMAKE_MODULE_PATH中提供“Findaslam_cv_common.cmake”，该项目要求CMake查找“aslam_cv_common”提供的包配置文件，但CMake没有找到。
> 
> 找不到“aslam_cv_common”提供的具有以下任何名称的包配置文件：
> 
> ```
> aslam_cv_commonConfig.cmake
> aslam_cv_common-config.cmake
> ```
> 
> 将安装前缀“aslam_cv_common”添加到CMAKE_PREFIX_PATH或将“aslam_cv_common_DIR”设置为包含上述文件之一的目录。如果“aslam_cv_common”提供了单独的开发包或 SDK，请确保已安装它。调用堆栈（最近的调用优先）：CMakeLists.txt：5 （catkin_simple）

aslam_ cv_ Common is one of the sub modules. You can find this file in the maplab directory, click on it, enter the Github page of this sub module, download the zip file, and then place it in the corresponding location. There are dozens of such sub modules in the maplab project.

MF-Ahmed, 2023-04-26:
Thanks, I have installed all the required sub-modules, yes there were  a lot of them. No during catkin build I'm getting the following error 

Errors     << aslam_cv_cameras:make /home/usr/data/catkin_ws/logs/aslam_cv_cameras/build.make.001.log                                         
In file included from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-unified-projection.h:4,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/src/camera-unified-projection.cc:3:
/home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
    7 | #include "aslam/common/id.pb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-3d-lidar.h:4,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/src/camera-3d-lidar.cc:1:
/home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
    7 | #include "aslam/common/id.pb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-pinhole.h:4,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/src/camera-pinhole.cc:4:
/home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
    7 | #include "aslam/common/id.pb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/aslam_cv_cameras.dir/build.make:102: CMakeFiles/aslam_cv_cameras.dir/src/camera-unified-projection.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/aslam_cv_cameras.dir/build.make:63: CMakeFiles/aslam_cv_cameras.dir/src/camera-3d-lidar.cc.o] Error 1
make[2]: *** [CMakeFiles/aslam_cv_cameras.dir/build.make:89: CMakeFiles/aslam_cv_cameras.dir/src/camera-pinhole.cc.o] Error 1
In file included from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-factory.h:10,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/src/camera-factory.cc:1:
/home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
    7 | #include "aslam/common/id.pb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/aslam_cv_cameras.dir/build.make:76: CMakeFiles/aslam_cv_cameras.dir/src/camera-factory.cc.o] Error 1
In file included from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/src/camera.cc:5:
/home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
    7 | #include "aslam/common/id.pb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12,
[... cut ...]
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-factory.h:10,
                 from /home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_cameras/src/ncamera.cc:6:
/home/usr/data/catkin_ws/src/maplab-master/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
    7 | #include "aslam/common/id.pb.h"

sandeshchand87, 2025-07-10:
Is there any solution for   CMake Error at /home/usr/data/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
By not providing "Findbrisk.cmake" in CMAKE_MODULE_PATH this project has
asked CMake to find a package configuration file provided by "brisk", but
CMake did not find one.
i am not able to install the  brisk package from github repo.