ethz-asl/maplab #380: sensor_calibration_file fromat ? All wiki tutorial can't  be used.

cshang412, 2023-03-27:
i am following the wiki steps by step, i found that all wiki demo are use the old format calibration file, i spend 2 days to install all packages at Ubuntu20.04 noetic, and spend 1 trying to figure out why this repo need so many dependencies and it topo-connection, then i found out i can't use the demo.

For Example :
This is the recommend way to launch rovioli.

 `#!/usr/bin/env bash
LOCALIZATION_MAP_OUTPUT=$1
ROSBAG=$2
NCAMERA_CALIBRATION="$ROVIO_CONFIG_DIR/ncamera-euroc.yaml"
IMU_PARAMETERS_MAPLAB="$ROVIO_CONFIG_DIR/imu-adis16488.yaml"
IMU_PARAMETERS_ROVIO="$ROVIO_CONFIG_DIR/imu-sigmas-rovio.yaml"
REST=$@

rosrun rovioli rovioli \
  --alsologtostderr=1 \
  --v=2 \
  --ncamera_calibration=$NCAMERA_CALIBRATION  \
  --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB \
  --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO \
  --datasource_type="rosbag" \
  --save_map_folder="$LOCALIZATION_MAP_OUTPUT" \
  --optimize_map_to_localization_map=false \
  --map_builder_save_image_as_resources=false \
  --datasource_rosbag=$ROSBAG $REST`

  But the launch file NEED sensor_calibration_file , the dataset of the bag does't has the sensor_calibration_file, it need re-merge i think, but there is no doc about how to update this .
  
  God i spend 3 days try to install and figure out this packages and how to launch the demos,  i think this is so unfriently for  develpers to make contribution, and the repo's dependencies is so complicated. i hope some one can clean this repo and prepare the dataset and wiki as soon as possible .

smauq, 2023-03-28:
So that you know, being friendly will usually also get you friendlier help. We just updated the code and working on both fixes and documentation.

If you want to pull the old version of maplab you can always go to the tags we have or the release section [here](https://github.com/ethz-asl/maplab/releases). I am updating the docs in this PR https://github.com/ethz-asl/maplab/pull/375, will try to have it in a few days on how to run everything.

As for just getting you started you can run the EuRoC example that the original wiki also showed how simply by doing `roslaunch maplab_node euroc-maplab-node-w-rovioli.launch` and then in a separate terminal doing `rosbag play ...` whichever sequence of EuRoC you want to run. The structure of the calibration files and launch file I would hope is pretty evident [here](https://github.com/ethz-asl/maplab/tree/master/applications/maplab-node/launch/euroc). If not you're always welcome to come with more questions here.

After the map creation the rest of the process hasn't changed much, just extra features.

goldbattle, 2023-08-14:
Example [map building mode](https://maplab.asl.ethz.ch/docs/master/pages/tutorials-rovioli/B_Running-ROVIOLI-in-VIO-mode.html#building-a-map-from-a-rosbag) with new configs:
```bash
rosrun rovioli rovioli \
    --alsologtostderr=1 \
    --v=2 \
    --sensor_calibration_file=$CALIBRATION  \
    --datasource_type="rosbag" \
    --save_map_folder="$LOCALIZATION_MAP_OUTPUT/" \
    --optimize_map_to_localization_map=false \
    --map_builder_save_image_as_resources=false \
    --datasource_rosbag=$ROSBAG $REST
```

Example of [running on a built map](https://maplab.asl.ethz.ch/docs/master/pages/tutorials-rovioli/C_Running-ROVIOLI-in-Localization-mode.html#running-rovioli-with-localization):
```bash
rosrun rovioli rovioli \
    --alsologtostderr=1 \
    --v=2 \
    --sensor_calibration_file=$CALIBRATION  \
    --publish_debug_markers  \
    --datasource_type="rosbag" \
    --optimize_map_to_localization_map=false \
    --vio_localization_map_folder=$LOCALIZATION_MAP_INPUT \
    --save_map_folder=$LOCALIZATION_MAP_OUTPUT \
    --map_builder_save_image_as_resources=false \
    --datasource_rosbag=$DATASET $REST
```

Would be good to update the [kalibr conversion tool](https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/exporters/kalibr_maplab_config) as it isn't very straightforward to get the new format (I ran into some issues with indenting correctly that the console "Failed to deserialize sensor!" errors where not that helpful to debug).