ethz-asl/maplab #392: "aslam/common/id.pb.h" not found

MF-Ahmed, 2023-04-26:
Hi,
 I have Ubunto 18.04 ROS Melodic during the compilation getting the following error

Errors     << aslam_cv_cameras:make /home/uzi/data/catkin_ws/logs/aslam_cv_cameras/build.make.001.log                                        
In file included from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12:0,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-3d-lidar.h:4,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_cameras/src/camera-3d-lidar.cc:1:
/home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
 #include "aslam/common/id.pb.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/aslam_cv_cameras.dir/src/camera-3d-lidar.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_common/include/aslam/common/unique-id.h:12:0,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_common/include/aslam/common/sensor.h:7,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera.h:16,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_cameras/include/aslam/cameras/camera-factory.h:10,
                 from /home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_cameras/src/camera-factory.cc:1:
/home/uzi/data/catkin_ws/src/aslam_cv2/aslam_cv_common/include/aslam/common/internal/unique-id.h:7:10: fatal error: aslam/common/id.pb.h: No such file or directory
 #include "aslam/common/id.pb.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/aslam_cv_cameras.dir/src/camera-factory.cc.o] Error 1
make[1]: *** [CMakeFiles/aslam_cv_cameras.dir/all] Error 2
make: *** [all] Error 2

smauq, 2023-04-26:
Hi,
A few quick questions:
- Is this a clean compile, or are you recompiling after changing something?
- Did you change anything in the protoc file?
- Do you have ccache installed?

AcamelQ, 2023-05-09:
i encounter this problem when i tried to compile it on ROS noetic Ubuntu 20.04. actually there isnt a directory called aslam/common, how can a fix it. i followed the setting procedure in your website.

smauq, 2023-05-13:
Ah, sorry, didn't see the issue earlier. So a clean compile?
- Did you follow [this](https://github.com/ethz-asl/maplab/blob/4b705604e583639dd7480fad9136223794b98749/docs/pages/installation/A_Installation-Ubuntu-20.04.md)?
- If you run `dpkg -l | grep protobuf` on the terminal, what do you get?

That file `aslam/common/id.pb.h` is automatically generated by google protobuf. So it means that for some reason protoc is not running. Maybe there's an issue with how the dependency is set up.

closed 2023-05-15 by MF-Ahmed

chenfusheng233, 2025-02-19:
@MF-Ahmed hi,How did you solve this problem?