ethz-asl/maplab #393: Compilation Error ROS Noetic, Ubunto 20.04

MF-Ahmed, 2023-05-15:
Hi, 
I am having the following error  wile compiling Maplab 2.0 on Ubunto2 0.4, ROS noetic 
Errors     << evaluation_tools:make /home/usr/data/maplab_ws/logs/evaluation_tools/build.make.002.log                                                                                                                                                                                    
/home/usr/data/maplab_ws/src/evaluation_tools/src/evaluation-data-collector.cc:14:42: error: ‘Random’ is not a member of ‘evaluation::internal::EvaluationDataCollectorImpl::SlotId’ {aka ‘aslam::NFramesId’}
   14 |     EvaluationDataCollectorImpl::SlotId::Random();
      |                                          ^~~~~~

λ › catkin config                                                                                                                                                                                                                                                           maplab_ws/src
----------------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/usr/data/maplab_ws
----------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/usr/data/maplab_ws/build
Devel Space:        [exists] /home/usr/data/maplab_ws/devel
Install Space:      [unused] /home/usr/data/maplab_ws/install
Log Space:          [exists] /home/usr/data/maplab_ws/logs
Source Space:       [exists] /home/usr/data/maplab_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
----------------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_FLAGS=-fdiagnostics-color
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
----------------------------------------------------------------------------------------------------
Workspace configuration appears valid.

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

MF-Ahmed, 2023-05-16:
Thanks 
I wen to https://github.com/ethz-asl/maplab/blob/4b705604e583639dd7480fad9136223794b98749/docs/pages/installation/A_Installation-Ubuntu-20.04.md
but I am stuck here for ages.

git clone git@github.com:ethz-asl/maplab.git --recursive                                                                                                                                                                                                                maplab_ws/src
Cloning into 'maplab'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

 That's why I tried to compile for the github repo directly .

smauq, 2023-05-17:
Do you have [a SSH key](https://docs.github.com/en/authentication/connecting-to-github-with-ssh/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent) in your github profile?

MF-Ahmed, 2023-05-17:
Thanks finally i manged to compile with the SSH key. 
Thanks