ethz-asl/maplab #394: Unable to open example VI map in maplab - missing vi_map/vi_map_metadata

ryanalex98, 2023-05-15:
Hello, I am just getting familiar with this repository. I have built maplab and am attempting to load one of the example VI maps into maplab (I have tried downloading vi-maps for CLA floor F, Zurich old town dataset, and EuRoC MH01). However, all of the maps are missing the /vi_map/vi_map_metadata file, and because of this I am encountering a serialization error when running "maplab load". Am I missing a step in the setup?

Error message:
```
vi-map-serialization.cc:288] Couldn't read the map metadata file.
```

Let me know if I can provide any other information. Thanks!

Some info:
     - Operating System: 64 bit, Ubuntu 20.04
     - ROS Noetic

smauq, 2023-05-17:
Old maps, I'm removing those links in the new wiki update. Best solution is to build your own map from EuRoC. If you have everything compiled just source the workspace. And then run:
`roslaunch maplab_node euroc-maplab-node-w-rovioli.launch`
And in parallel rosbag play one of the EuRoC bags from [here](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets).

After the bag is finished the map should save automatically to `/tmp/euroc/` and you can continue with the maplab console part. Let me know if that works.

ryanalex98, 2023-05-17:
Hi, thanks for the reply. I tried doing this and the result was an empty /resources directory located in /tmp/euroc. It looks like the maplab node shut down prematurely due to no odometry data received. In the Rviz visualization, the only thing that happens when rosbag is played is the TF is updated. Here's a block of code from the terminal window right when the node shuts down.

```
I20230517 10:43:43.130059  4015 rovioli-app.cc:170] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
DataPublisherFlow             LOCALIZATION_RESULT           ca9a..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           a3da..0000                    0                             
DataPublisherFlow             ROVIO_ESTIMATES               dba0..0000                    0                             
RovioFlow                     ODOMETRY_MEASUREMENTS         dc94..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            3b74..0000                    0                             
RovioFlow                     IMU_MEASUREMENTS              1cb7..0000                    0                             
-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1
W20230517 10:43:47.031739  4077 synchronizer.cc:949] [MaplabNode-Synchronizer] No odometry messages have been received in the last 5 seconds.
W20230517 10:43:47.031862  4077 synchronizer.cc:960] [MaplabNode-Synchronizer] Check if the topic is properly set in the sensor calibration file.
I20230517 10:43:51.785070  4052 synchronizer.cc:353] [MaplabNode-Synchronizer] Received first odometry message!
I20230517 10:43:53.131938  4015 rovioli-app.cc:170] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
DataPublisherFlow             LOCALIZATION_RESULT           ca9a..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           a3da..0000                    0                             
DataPublisherFlow             ROVIO_ESTIMATES               dba0..0000                    0                             
RovioFlow                     ODOMETRY_MEASUREMENTS         dc94..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            3b74..0000                    389                           
RovioFlow                     IMU_MEASUREMENTS              1cb7..0000                    1944                          
I20230517 10:43:57.009605  4016 maplab-ros-node-app.cc:49] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
FeatureTracking               SYNCED_NFRAMES                f98a..0000                    0                             
DataPublisher                 RAW_VIMAP                     918a..0000                    0                             
MapBuilder                    SYNCED_LOOP_CLOSURE           408f..0000                    0                             
MapBuilder                    SYNCED_WHEEL_ODOMETRY         a1e1..0000                    0                             
Synchronizer                  EXTERNAL_FEATURES             ef5c..0000                    0                             
DataPublisher                 MAP_UPDATES                   cd37..0000                    0                             
DataPublisher                 ODOMETRY_ESTIMATES            a7b7..0000                    0                             
MapBuilder                    MAP_UPDATES                   5b4b..0000                    0                             
MapBuilder                    SYNCED_LIDAR_MEASUREMENTS     b9de..0000                    0                             
Synchronizer                  IMAGE_MEASUREMENTS            40b7..0000                    0                             
MapBuilder                    SYNCED_POINTCLOUD_MAP         9f00..0000                    0                             
MapBuilder                    SYNCED_ABSOLUTE_6DOF          2552..0000                    0                             
MapBuilder                    TRACKED_NFRAMES               e875..0000                    0                             
MapBuilder                    SYNCED_EXTERNAL_FEATURES      1f3b..0000                    0                             
Synchronizer                  ODOMETRY_ESTIMATES            11db..0000                    0                             
Synchronizer                  IMU_MEASUREMENTS              aae2..0000                    0                             
I20230517 10:44:03.135663  4015 rovioli-app.cc:170] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
DataPublisherFlow             LOCALIZATION_RESULT           ca9a..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           a3da..0000                    0                             
DataPublisherFlow             ROVIO_ESTIMATES               dba0..0000                    0                             
[... cut ...]
RovioFlow                     IMU_MEASUREMENTS              1cb7..0000                    3914                          
W20230517 10:44:12.033022  4077 synchronizer.cc:949] [MaplabNode-Synchronizer] No odometry messages have been received in the last 5 seconds.
W20230517 10:44:12.033246  4077 synchronizer.cc:954] [MaplabNode-Synchronizer] Either the data sources has stopped publishing odometry estimates or there has been a data drop. Triggering shutdown!
I20230517 10:44:13.009806  4016 maplab-ros-node.cc:199] [MaplabROSNode] Shutting down...
```

smauq, 2023-05-20:
The topics seem fine. What seems to be the problem is that rovioli seems to diverge or have problems as it's just buffering a lot of data internally. You can see from these lines:
```
RovioFlow                     IMAGE_MEASUREMENTS            3b74..0000                    784                           
RovioFlow                     IMU_MEASUREMENTS              1cb7..0000                    3914  
```
Are you running this on an extremely slow PC? Or playing the bag at a high speed? If you set this to true, you can visualize what ROVIO is doing https://github.com/ethz-asl/maplab/blob/ab0d958201f8bfead8fa9f66d464be229c5893b1/applications/maplab-node/launch/euroc/euroc-maplab-node-w-rovioli.launch#L21 and let me know? I just quickly rechecked, on my machine the launch file works without problems (same build Ubuntu 20.04 / Noetic).

ryanalex98, 2023-05-20:
Thanks for the suggestion. I was playing back the rosbag at the normal rate, and it was crashing before receiving its first odometry message. I had some luck playing the rosbag at 0.1x speed, with the console out confirming that the first odometry message had been received, but this didn't carry on as MaplabNode-Synchronizer quickly shut everything in the same manner after not receiving an odometry message for 5 seconds. There was a warning message related to the canberra-gtk-module failing to load after the first odometry message was received, perhaps that could be the cause? 

Having the rovio frame visualization tracker enabled to see the first bits of progress was a nice touch. Fyi, after the MaplabNode-Synchronizer shut everything down, the tracker continued updating. I think this is expected behavior but wanted to share just in case.

Regarding my CPU, I am running Ubuntu 20.04 on a virtual machine. It is configured with 12GB of RAM and 8 cores. I ran the process with System Monitor in the background and neither the memory nor the CPUs seemed to be maxing out when the node shuts down, so I don't think its performance-related, but I could be wrong. Which branch are you on? I am using the latest master branch.

Thank you for your help!

```
-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1
W20230520 04:19:03.086707 37877 synchronizer.cc:949] [MaplabNode-Synchronizer] No odometry messages have been received in the last 5 seconds.
W20230520 04:19:03.087311 37877 synchronizer.cc:960] [MaplabNode-Synchronizer] Check if the topic is properly set in the sensor calibration file.
Gtk-Message: 04:19:04.572: Failed to load module "canberra-gtk-module"
I20230520 04:19:04.665557 37852 synchronizer.cc:353] [MaplabNode-Synchronizer] Received first odometry message!
I20230520 04:19:06.377945 37817 rovioli-app.cc:170] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
DataPublisherFlow             ROVIO_ESTIMATES               154f..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           5964..0000                    0                             
DataPublisherFlow             LOCALIZATION_RESULT           19a7..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            b49f..0000                    23                            
RovioFlow                     ODOMETRY_MEASUREMENTS         3509..0000                    0                             
RovioFlow                     IMU_MEASUREMENTS              a0ff..0000                    114                           
I20230520 04:19:13.100283 37818 maplab-ros-node-app.cc:49] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
Synchronizer                  EXTERNAL_FEATURES             3dec..0000                    0                             
DataPublisher                 MAP_UPDATES                   0185..0000                    0                             
DataPublisher                 RAW_VIMAP                     1c1a..0000                    0                             
DataPublisher                 ODOMETRY_ESTIMATES            f264..0000                    0                             
MapBuilder                    SYNCED_LIDAR_MEASUREMENTS     8b9c..0000                    0                             
MapBuilder                    MAP_UPDATES                   d7f7..0000                    0                             
Synchronizer                  IMAGE_MEASUREMENTS            5bec..0000                    0                             
MapBuilder                    SYNCED_ABSOLUTE_6DOF          d779..0000                    0                             
MapBuilder                    SYNCED_WHEEL_ODOMETRY         8b68..0000                    0                             
MapBuilder                    SYNCED_LOOP_CLOSURE           5f3a..0000                    0                             
MapBuilder                    SYNCED_POINTCLOUD_MAP         3c95..0000                    0                             
MapBuilder                    TRACKED_NFRAMES               2fc6..0000                    0                             
FeatureTracking               SYNCED_NFRAMES                e7e6..0000                    0                             
Synchronizer                  ODOMETRY_ESTIMATES            ed6a..0000                    0                             
MapBuilder                    SYNCED_EXTERNAL_FEATURES      a79c..0000                    0                             
Synchronizer                  IMU_MEASUREMENTS              9bba..0000                    0                             
I20230520 04:19:16.380450 37817 rovioli-app.cc:170] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
DataPublisherFlow             ROVIO_ESTIMATES               154f..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           5964..0000                    0                             
DataPublisherFlow             LOCALIZATION_RESULT           19a7..0000                    0                             
[... cut ...]
RovioFlow                     LOCALIZATION_RESULT           5964..0000                    0                             
DataPublisherFlow             LOCALIZATION_RESULT           19a7..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            b49f..0000                    89                            
RovioFlow                     ODOMETRY_MEASUREMENTS         3509..0000                    0                             
RovioFlow                     IMU_MEASUREMENTS              a0ff..0000                    444
```

smauq, 2023-05-22:
Hi! So then the problem isn't in maplab per se, but more about your machine setup. Maplab shuts down automatically after a more than 5 second message interruption, that's normal. It seems rovioli, which is the odometry front end, is going very very slowly as it's just buffering stuff. That canberra-gtk-module failing error does not appear for me. Your virtual machine just seems to not be doing its job very well. If you need to run this on Windows can you just use WSL? I've had people running that without too many problems before.

ryanalex98, 2023-05-22:
Ok, good to know. I will try using a different configuration. Thanks!