ethz-asl/maplab #408: undefined reference to `void backend::addLabelToPointCloud<resources::VoxbloxColorPointCloud>(unsigned int, unsigned long, resources::VoxbloxColorPointCloud*)'

Mysterium-sch, 2024-01-24:
I am trying to run maplab 2.0 on 20.04 noetic. To install the python packages I had to change python to python3. When i go to build maplab the maplab console fails due to the following error: undefined reference to `void backend::addLabelToPointCloud<resources::VoxbloxColorPointCloud>(unsigned int, unsigned long, resources::VoxbloxColorPointCloud*)'

I was wondering if anyone else has seen this issue. I have provided my catkin config below.

-----------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /maplab_ws
-----------------------------------------------
Build Space:        [exists] /maplab_ws/build
Devel Space:        [exists] /maplab_ws/devel
Install Space:      [unused] /maplab_ws/install
Log Space:          [exists] /maplab_ws/logs
Source Space:       [exists] /maplab_ws/src
DESTDIR:            [unused] None
-----------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-----------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------

yuxuanzhou97, 2024-12-02:
I meet this problem too. Do you have some idea how to solve it?

yuxuanzhou97, 2024-12-02:
I temporarily skip this problem by comment "addLabelToPointCloud(label, point_idx, output_cloud);" in "src/maplab/backend/map-resources/include/map-resources/resource-conversion-inl.h" 😂

LiShiwei18, 2025-09-15:
Please refer on ubuntu 20.04:
https://github.com/ethz-asl/maplab/blob/4b705604e583639dd7480fad9136223794b98749/docs/pages/installation/A_Installation-Ubuntu-20.04.md
Instead of: 
https://github.com/ethz-asl/maplab/blob/master/docs/pages/installation/A_Installation-Ubuntu.md
The first one will use the master branch, the second one will use the develop branch, the above error will appear on the develop branch