ethz-asl/maplab #414: try maplab2.0 rovioli failed

grn213331, 2024-09-26:
when I  try rosrun rovioli tutorial_euroc save_folder MH_01_easy.bag
there are errors
ERROR: unknown command line flag 'imu_parameters_maplab'
ERROR: unknown command line flag 'ncamera_calibration'

so I  find the command is based on maplab1.0 , but I have built maplab2.0
About maplab2.0,  what is the command?

my build branch is develop.

W-upwind, 2025-01-17:
So, may I ask if the issue has been resolved now?

maikelborys, 2025-07-10:
Hello firends! Im facing same error. Thanks! 

ERROR: unknown command line flag 'imu_parameters_maplab'
ERROR: unknown command line flag 'imu_parameters_rovio'
ERROR: unknown command line flag 'ncamera_calibration'